ROBOTIS-GIT/turtlebot3_simulations #196: Integration for Gazebo Sim 7 (Garden)

brean, 2022-11-26:
Hey!
based on the feature-ign-gazebo-branch I created a version that works with the latest versions of gz_ros2_control, ros_gz (ros2 branch) and gz-transport (gz-transport12 branch) that works nicely with Gazebo Sim 7 (Garden): https://github.com/brean/turtlebot3_simulations/tree/brean/feature-gz-sim_garden .
You can see it in action using docker here: https://github.com/brean/ros2-turtlebot3-gazebo-docker
It was relatively easy, most if just simple search-and-replace.
Maybe I can create a merge request for it but maybe you guys want to wait until its all a bit more stable and create one yourself?
With the main-branch being ~150 commits away from master it maybe also makes sense to take a deeper look into creating a clean version?

ahcorde, 2023-06-16:
I made this work long time ago, but it was not merge in `galactic-devel`, just before merge it was retargeted to `feature-ign_gazebo`.

@ROBOTIS-Will why did you not merge this in `galactic` ? is there any issue about giving support for the "new" Gazebo ?

I'm happy to put some time on it, but if the contribution will be merge in `rolling`, `iron` or `humble`.

j-rivero, 2023-10-25:
> I made this work long time ago, but it was not merge in `galactic-devel`, just before merge it was retargeted to `feature-ign_gazebo`.

Different users are asking for the support in the new Gazebo. For reference, the @ahcorde  PR was https://github.com/ROBOTIS-GIT/turtlebot3_simulations/pull/180.

From the Gazebo team, we are happy to collaborate to push that branch into the stable if you need any help.

azeey, 2023-12-22:
Any progress on this?
As you may know, Gazebo-classic (aka Gazebo11, see [Terminology](https://gazebosim.org/about#:~:text=Terminology)) is no longer being actively developed  and will not be available on ROS Jazzy ([REP 2000](https://www.ros.org/reps/rep-2000.html#jazzy-jalisco-may-2024-may-2029)).

chan-1207, 2025-06-10:
Sorry for the late reply. Our Open Platform Team was newly established in 2025, and we have started maintaining the previously inactive open platform. An update to Gazebo Sim 8 has been released for ROS 2 Jazzy, check it out.

GyuH13, 2025-06-10:
With support for the Jazzy version, Gazebo Harmonic is now also supported. Thank you. I will now close this issue.