ROBOTIS-GIT/turtlebot3_simulations #198: Issue launch empty world OR any other world

KaeYan, 2022-12-30:
This is the steps i did after fresh install of Ubuntu 22.04

I followed the [instructions](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html) to install ROS 2 humble, the `talker` and `listener` is able to work.

Next, this is the command i run to install `turtlebot3_simulations`
`$ sudo apt-get install ros-humble-gazebo-*`
`$ sudo apt install ros-humble-cartographer`
`$ sudo apt install ros-humble-cartographer-ros`
`$ sudo apt install ros-humble-navigation2`
`$ sudo apt install ros-humble-nav2-bringup`
`$ sudo apt install ros-humble-dynamixel-sdk`
`$ sudo apt install ros-humble-turtlebot3-msgs`
`$ sudo apt install ros-humble-turtlebot3`
`$ mkdir -p ~/turtlebot3_ws/src/`
`$ git clone -b humble-devel https://github.com/ROBOTIS-GIT/turtlebot3_simulations.git`

Then this is the command to build. It did success building all 3 packages.
`$ source /opt/ros/humble/setup.bash`
`$ cd ~/turtlebot3_ws && colcon build --symlink-install` 

Then this is the command to run the simulations. 
`$ export TURTLEBOT3_MODEL=waffle`
`$ ros2 launch turtlebot3_gazebo empty_world.launch.py`

When running, i encountered the following error:
```
[INFO] [launch]: All log files can be found below /home/kaeyan/.ros/log/2022-12-30-22-10-06-854162-kaeyan-70995
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle.urdf
urdf_file_name : turtlebot3_waffle.urdf
[INFO] [gzserver-1]: process started with pid [70996]
[INFO] [gzclient-2]: process started with pid [70998]
[INFO] [robot_state_publisher-3]: process started with pid [71000]
[INFO] [spawn_entity.py-4]: process started with pid [71002]
[robot_state_publisher-3] Link base_link had 7 children
[robot_state_publisher-3] Link camera_link had 2 children
[robot_state_publisher-3] Link camera_depth_frame had 1 children
[robot_state_publisher-3] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-3] Link camera_rgb_frame had 1 children
[robot_state_publisher-3] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-3] Link caster_back_left_link had 0 children
[robot_state_publisher-3] Link caster_back_right_link had 0 children
[robot_state_publisher-3] Link imu_link had 0 children
[robot_state_publisher-3] Link base_scan had 0 children
[robot_state_publisher-3] Link wheel_left_link had 0 children
[robot_state_publisher-3] Link wheel_right_link had 0 children
[robot_state_publisher-3] [INFO] [1672413007.697509945] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1672413007.697578471] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1672413007.697587904] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1672413007.697593326] [robot_state_publisher]: got segment camera_depth_frame
[robot_state_publisher-3] [INFO] [1672413007.697597787] [robot_state_publisher]: got segment camera_depth_optical_frame
[robot_state_publisher-3] [INFO] [1672413007.697602384] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1672413007.697606882] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-3] [INFO] [1672413007.697611339] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-3] [INFO] [1672413007.697615829] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-3] [INFO] [1672413007.697620149] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-3] [INFO] [1672413007.697624663] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1672413007.697629118] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1672413007.697633531] [robot_state_publisher]: got segment wheel_right_link
[spawn_entity.py-4] [INFO] [1672413007.965715422] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1672413007.965973847] [spawn_entity]: Loading entity XML from file /home/kaeyan/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_waffle/model.sdf
[spawn_entity.py-4] [INFO] [1672413007.966544342] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1672413007.966761544] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [ERROR] [1672413038.028179909] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-4] [ERROR] [1672413038.029427118] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-4]: process has died [pid 71002, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity waffle -file /home/kaeyan/turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_waffle/model.sdf -x 0.0 -y 0.0 -z 0.01 --ros-args'].
```
I even tried to export the `GAZEBO_MODEL_PATH` by running the following, same error still occur:
`$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models`

KaeYan, 2022-12-30:
Also, does this simulations depends on graphic card driver?

The following is the output after the command `$ sudo lshw -c video`. Not sure if this helps. 

```
  *-display                 
       description: VGA compatible controller
       product: GA104M [GeForce RTX 3070 Mobile / Max-Q]
       vendor: NVIDIA Corporation
       physical id: 0
       bus info: pci@0000:01:00.0
       version: a1
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress vga_controller bus_master cap_list rom
       configuration: driver=nvidia latency=0
       resources: iomemory:400-3ff iomemory:420-41f irq:183 memory:a3000000-a3ffffff memory:4000000000-41ffffffff memory:4200000000-4201ffffff ioport:3000(size=128) memory:c0000-dffff
  *-graphics
       product: EFI VGA
       physical id: 2
       logical name: /dev/fb0
       capabilities: fb
       configuration: depth=32 resolution=1920,1080
```

thehummingbird, 2023-02-23:
I have the same issue. Did you find a solution to this?

dbortoluzzi, 2023-03-31:
Me too. Same issue.

daylight0, 2023-06-30:
Me too,  I'm trying to resolve it.

Astaxuqichao, 2023-07-04:
I encounter the same issue.
But I find if turn off all network(wired or wifi),  the world open.
I guess gazebo11 is in trouble with network(gazebo 9 is normal)
Hope it can help you.

WorstCodeWay, 2025-03-06:
Me too, more than two years later...

WorstCodeWay, 2025-03-06:
Finally, I solve it by (on my situation) exporting path of gazebo path in my running terminal with
```
export GAZEBO_MODEL_PATH=/usr/share/gazebo-11/models
```
And I have added this exporting command into ~/.bashrc to export it every time a terminal opened.

Putting here in case anyone meet this problem.

chan-1207, 2025-06-10:
Hello, We apologize for the delay in responding.

Make sure you've sourced environment variables.
```source /usr/share/gazebo/setup.bash```
This serves to set environment variables in the current shell session that are needed when using the Gazebo simulator.

If this does not resolve your issue, please open an additional issue and we will close this one. 
Thank you for your contribution.

closed 2025-06-10 by GyuH13