ROBOTIS-GIT/turtlebot3_simulations #199: ros2 launch turtlebot3_gazebo empty_world.launch.py error

jsb9945, 2023-02-01:
Hello!
My environment is ros2 foxy ubuntu 20.04.
I executed this command.

ros2 launch turtlebot3_gazebo empty_world.launch.py

then, i got this error

[INFO] [launch]: All log files can be found below /home/csrl/.ros/log/2023-02-01-12-30-26-169214-csrl-System-Product-Name-61039
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/csrl/aws_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/launch/empty_world.launch.py", line 28, in <module>
    TURTLEBOT3_MODEL = os.environ['TURTLEBOT3_MODEL']
  File "/usr/lib/python3.8/os.py", line 675, in __getitem__
    raise KeyError(key) from None
KeyError: 'TURTLEBOT3_MODEL'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [py]: 'TURTLEBOT3_MODEL'

What's the problem? please help me

saiaravind19, 2023-09-13:
The mode which needs to be spawned is not provided so is the error.

1. To spawn turtlebot use 

` export TURTLEBOT3_MODEL=burger`
`ros2 launch turtlebot3_gazebo empty_world.launch.py`

2.To spawn waffle use 
` export TURTLEBOT3_MODEL=waffle`
`ros2 launch turtlebot3_gazebo empty_world.launch.py`

In case you always use turtlebot then use the below command
`echo  "export TURTLEBOT3_MODEL=burger" >> ~/.bashrc`

chan-1207, 2025-06-10:
Hello, We apologize for the delay in responding.

This is caused by the TURTLEBOT3_MODEL environment variable not being specified. Please refer to the answer above.

If this does not resolve your issue, please open an additional issue and we will close this one.
Thank you for your contribution.