ROBOTIS-GIT/turtlebot3_simulations #206: turtlebot3_world.launch local_costmap warnings

berreuphoria, 2023-10-24:
Hello I'm new on ROS. i just created a catkin workspace and i installed turtlebot3 simulations and turtlebot3 from turtlebot3's website. i literally do nothing but these. But i have these warnings. How can i fix this?
Here are the commands i wrote on terminal: 
`    roslaunch turtlebot3_gazebo turtlebot3_world.launch`
`    roslaunch turtlebot3_navigation turtlebot3_navigation.launch`
These  are the warnings:
    **[ WARN] [1698154732.099839494, 7.086000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist. canTransform: source_frame base_footprint does not exist.. canTransform returned after 7.086 timeout was 0.1.
    [ WARN] [1698154732.716332026, 7.681000000]: global_costmap: Parameter "plugins" not provided, loading pre-Hydro parameters
    [ WARN] [1698154733.188402789, 8.128000000]: local_costmap: Parameter "plugins" not provided, loading pre-Hydro parameters**

These are my local_costmap_params.yaml and global_costmap_params.yaml files: 
 local_costmap_params.yaml 
```
local_costmap:
  global_frame: odom
  robot_base_frame: base_footprint

  update_frequency: 10.0
  publish_frequency: 10.0
  transform_tolerance: 0.5  

  static_map: false  
  rolling_window: true
  width: 3
  height: 3
  resolution: 0.05
```
global_costmap_params.yaml 
```
global_costmap:
  global_frame: map
  robot_base_frame: base_footprint

  update_frequency: 10.0
  publish_frequency: 10.0
  transform_tolerance: 0.5

  static_map: true
```

TZECHIN6, 2025-01-07:
Have you created the map? Or you are trying to run SLAM?

chan-1207, 2025-06-10:
Hello.

Warning that the plugins parameter is missing from the config file of the costmap_2d package, which is attempting to load parameters in an outdated (pre-Hydro) way. 

It works fine, but we recommend using the humble version.

sunghowoo, 2025-06-20:
We are currently supporting ROS 2 Humble, and this issue is quite outdated, so we will go ahead and close it. If the issue still persists, please feel free to open a new one. Apologies for the late response.

closed 2025-06-20 by sunghowoo