ROBOTIS-GIT/turtlebot3_simulations #207: Gazebo simulation not launching in humble

nandu-k01, 2023-11-15:
Updated to ros2 humble and ubuntu 22.04
This is the error
```ros2 launch turtlebot3_gazebo empty_world.launch.py 
[INFO] [launch]: All log files can be found below /home/zorx/.ros/log/2023-11-15-19-13-16-666336-zorx-48737
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_waffle_pi.urdf
urdf_file_name : turtlebot3_waffle_pi.urdf
[INFO] [gzserver-1]: process started with pid [48738]
[INFO] [gzclient-2]: process started with pid [48740]
[INFO] [robot_state_publisher-3]: process started with pid [48742]
[INFO] [spawn_entity.py-4]: process started with pid [48744]
[robot_state_publisher-3] [INFO] [1700055797.383270144] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1700055797.383338670] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1700055797.383343727] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1700055797.383346669] [robot_state_publisher]: got segment camera_link
[robot_state_publisher-3] [INFO] [1700055797.383349388] [robot_state_publisher]: got segment camera_rgb_frame
[robot_state_publisher-3] [INFO] [1700055797.383352226] [robot_state_publisher]: got segment camera_rgb_optical_frame
[robot_state_publisher-3] [INFO] [1700055797.383355080] [robot_state_publisher]: got segment caster_back_left_link
[robot_state_publisher-3] [INFO] [1700055797.383357847] [robot_state_publisher]: got segment caster_back_right_link
[robot_state_publisher-3] [INFO] [1700055797.383360588] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1700055797.383363294] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1700055797.383366170] [robot_state_publisher]: got segment wheel_right_link
[spawn_entity.py-4] [INFO] [1700055797.608546486] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1700055797.608755151] [spawn_entity]: Loading entity XML from file /home/zorx/Desktop/tb3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_waffle_pi/model.sdf
[spawn_entity.py-4] [INFO] [1700055797.609309410] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1700055797.609487792] [spawn_entity]: Waiting for service /spawn_entity
[spawn_entity.py-4] [ERROR] [1700055827.665935221] [spawn_entity]: Service %s/spawn_entity unavailable. Was Gazebo started with GazeboRosFactory?
[spawn_entity.py-4] [ERROR] [1700055827.666978495] [spawn_entity]: Spawn service failed. Exiting.
[ERROR] [spawn_entity.py-4]: process has died [pid 48744, exit code 1, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity waffle_pi -file /home/zorx/Desktop/tb3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_waffle_pi/model.sdf -x 0.0 -y 0.0 -z 0.01 --ros-args'].
```

These are the env variables in my .bashrc

```
source /opt/ros/humble/setup.bash
export TURTLEBOT3_MODEL=waffle_pi
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
```

How to fix this??

closed 2023-11-17 by nandu-k01

RugvedKatole, 2024-01-20:
Were you able to fix this? I am facing the same problem.

victorcarvesk, 2024-03-09:
> Were you able to fix this? I am facing the same problem.

For some reason, the maintainer wrote code that calls Gazebo without the libraries necessary for ROS integration.
To solve this problem, you need to modify how the launch file calls Gazebo. Remove the calls to `gzserver` and `gzclient`, and replace them with a single call to `Gazebo`. This will ensure ROS launches it with the necessary libraries (one of these libraries is the factory library, which provides essential services like the spawn service).

[image]

[image]

A corrected version of the file/package can be found [here](https://github.com/victorcarvesk/turtlebot3_simulations).