ROBOTIS-GIT/turtlebot3_simulations #211: [ERROR] [gzserver-1]: process has died [pid 10000, exit code 255]

MattiaDeiRossi, 2023-12-20:
Hi all, 

I'm using ros2 humble.
Launching the command `ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py` I have the following error
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
[INFO] [launch]: All log files can be found below /home/vscode/.ros/log/2023-12-20-15-23-29-348982-debian-9982
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gzserver-1]: process started with pid [10000]
[INFO] [gzclient-2]: process started with pid [10002]
[INFO] [robot_state_publisher-3]: process started with pid [10004]
[INFO] [spawn_entity.py-4]: process started with pid [10006]
[robot_state_publisher-3] [INFO] [1703085810.375105888] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-3] [INFO] [1703085810.375199876] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1703085810.375212496] [robot_state_publisher]: got segment base_scan
[robot_state_publisher-3] [INFO] [1703085810.375221894] [robot_state_publisher]: got segment caster_back_link
[robot_state_publisher-3] [INFO] [1703085810.375229212] [robot_state_publisher]: got segment imu_link
[robot_state_publisher-3] [INFO] [1703085810.375237677] [robot_state_publisher]: got segment wheel_left_link
[robot_state_publisher-3] [INFO] [1703085810.375246384] [robot_state_publisher]: got segment wheel_right_link
[ERROR] [gzserver-1]: process has died [pid 10000, exit code 255, cmd 'gzserver /workspaces/ros-2/turtlebot3/install/turtlebot3_gazebo/share/turtlebot3_gazebo/worlds/turtlebot3_world.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].
[spawn_entity.py-4] [INFO] [1703085810.751628654] [spawn_entity]: Spawn Entity started
[spawn_entity.py-4] [INFO] [1703085810.752004276] [spawn_entity]: Loading entity XML from file /workspaces/ros-2/turtlebot3/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf
[spawn_entity.py-4] [INFO] [1703085810.752832807] [spawn_entity]: Waiting for service /spawn_entity, timeout = 30
[spawn_entity.py-4] [INFO] [1703085810.753139190] [spawn_entity]: Waiting for service /spawn_entity
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-3] [INFO] [1703085813.785133604] [rclcpp]: signal_handler(signum=2)
[spawn_entity.py-4] Traceback (most recent call last):
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 370, in <module>
[spawn_entity.py-4]     main()
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 365, in main
[spawn_entity.py-4]     exit_code = spawn_entity_node.run()
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 228, in run
[spawn_entity.py-4]     success = self._spawn_entity(entity_xml, initial_pose, spawn_service_timeout)
[spawn_entity.py-4]   File "/opt/ros/humble/lib/gazebo_ros/spawn_entity.py", line 276, in _spawn_entity
[spawn_entity.py-4]     if client.wait_for_service(timeout_sec=timeout):
[spawn_entity.py-4]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 177, in wait_for_service
[spawn_entity.py-4]     time.sleep(sleep_time)
[spawn_entity.py-4] KeyboardInterrupt
[ERROR] [gzclient-2]: process has died [pid 10002, exit code -2, cmd 'gzclient'].
[ERROR] [spawn_entity.py-4]: process has died [pid 10006, exit code -2, cmd '/opt/ros/humble/lib/gazebo_ros/spawn_entity.py -entity burger -file /workspaces/ros-2/turtlebot3/install/turtlebot3_gazebo/share/turtlebot3_gazebo/models/turtlebot3_burger/model.sdf -x -2.0 -y -0.5 -z 0.01 --ros-args'].
[INFO] [robot_state_publisher-3]: process has finished cleanly [pid 10004]
```

Have you got any suggestions how to solve it?

fatemeh-mohseni-AI, 2024-02-13:
killall gzserver
killall gzclient

delete ros2_ws and recreate it . 
then :
colcon build

retry. It might be resolved

Tanishq30052002, 2024-06-11:
doesnt work

manavthakkar, 2024-06-20:
It worked for me

Tanishq30052002, 2024-06-20:
for me source gazebo worked

Tanishq30052002, 2024-06-27:
> I still have the problem!
> Is there a chance that you might have fixed it?!

source /usr/share/gazebo/setup.sh

Try adding the same to your 
.bashrc

sathvik0320, 2024-08-10:
yes using source /usr/share/gazebo/setup.sh 
solved my error 
[ERROR] [gzserver-1]: process has died [pid 27337, exit code 255, cmd 'gzserver /home/sathvik/4wheel/install/control/share/control/model/empty.world -slibgazebo_ros_init.so -slibgazebo_ros_factory.so -slibgazebo_ros_force_system.so'].

offgrid88, 2024-08-24:
It is sourcing source /usr/share/gazebo/setup.sh most of the time or a hanging gz client or server 
anyone can add an alias 
alias killgazebo='killall gzserver ;killall gzclient'
to bashrc and make sure gazebo is killed

PAyush15, 2024-12-04:
> killall gzserver killall gzclient
> 
> delete ros2_ws and recreate it . then : colcon build
> 
> retry. It might be resolved

Thanks! That solved the issue for me.

sunghowoo, 2025-02-04:
This issue appears to be resolved, so I will close it.

closed 2025-02-04 by sunghowoo