ROBOTIS-GIT/turtlebot3_simulations #213: CMake Error in turtlebot3_gazeboConfig.cmake - Unable to find /usr/include/sdformat-9.8

LuMingJun62511, 2023-12-26:
### Issue Description

I encountered a CMake error while building the turtlebot3_gazebo package. The error message suggests that the project specifies '/usr/include/sdformat-9.8' as an include directory, but it is not found. The error occurs at line 113 in turtlebot3_gazeboConfig.cmake.

### Steps to Reproduce

1. Clone the turtlebot3_simulations repository.
2. Build the workspace using catkin_make.
3. Observe the reported CMake error.

### Error Logs

I have attached the relevant portions of CMakeOutput.log and CMakeError.log to this issue for reference.

### System Information

- ROS Version: noetic
- Operating System: Linux

### Additional Context

I checked the project's issue tracker and did not find a similar report. Please let me know if further information is needed to address this issue.

frienkie, 2024-12-03:
The most possible situation is that you update the sdformat package to 9.10. Noetic can only support 9.8 version.
use
`ls /usr/include/sdformat-9.8
` 
to check if 9.8 is still there. If not ,please use
`sudo aptitude install libsdformat9-dev=9.8.0-1~focal
`
to install or downgrade it to 9.8.

chan-1207, 2025-06-10:
This issue is an error that occurs when building or running the TurtleBot3 simulation package 
CMake cannot find the file (folder) `/usr/include/sdformat-9.8`.

### Install sdformat and related packages

You need to install sdformat (specifically 9.8 or later) for your system.

```bash
sudo apt update
sudo apt install libsdformat9-dev
```

---
If this does not resolve your issue, please open an additional issue and we will close this one.
Thank you for your contribution.

sunghowoo, 2025-06-20:
As there has been no recent activity, we will close this issue for now. If the issue still exists, please feel free to reopen it.