ROBOTIS-GIT/turtlebot3_simulations #221: Clarification on Protobuf dependency for turtlebot3_simulations

jorrgut, 2024-07-18:
Hi,
I am trying to compile the `turtlebot3_simulations` package on Ubuntu 22.04 with ROS 2 Humble and encountered an issue related to Protobuf. Could you please confirm if there is a specific version of Protobuf required for this package? If so, could you provide guidance on which version to use?

Thank you!

Best regards,
Jorge Gutierrez

yun-goon, 2024-12-05:
Hi,
Unfortunately, we do not officially support specific Protobuf versions for turtlebot3_simulations. 
If you encounter issues, feel free to share the details!
Thank you.

GyuH13, 2025-01-20:
This issue will be closed due to prolonged inactivity. If the problem persists, feel free to reopen it.

closed 2025-01-20 by GyuH13